adafruit/Adafruit_CCS811 #4: ESP8266 and CCS811_DRIVE_MODE_250MS. Crash! :(

Kameeno, 2018-02-06:
Hi! found a problem whit the example sketch on the ESP8266, i think is probably the Clock stretch when i run on mode 1s or 250ms, about what i edited on the library was just on the .cpp file the Wire.begin(); to my Wire.begin(2,14); and the 	setDriveMode on setDriveMode(CCS811_DRIVE_MODE_250MS);

she start to take 5,6,or some measures.. but after she going on crash and restart to rebooting (and failing the reboot) , someone he found this kind of issue? 

it's just whit the esp, it's working so good on teensy or other boards

ladyada, 2018-02-06:
yeah its likely either clock stretching or you need more yields() to keep the WDT from firing!

LucAce, 2018-02-16:
v2.3.0 of the ESP8266 Core for Arduino library has an issue with keeping the application set i2c clock stretching value.  The issue is that whenever begin() is called the default delay is set, even if you have previously set it to a larger value.  So it is likely if you have multiple begin() calls (for example if you have multiple sensors on I2C) you will need to call setClockStretchLimit(TIMEOUT) after each begin() call to work around it.  v2.4.0 of the ESP8266 Core for Arduino library is supposed to fix it.

https://github.com/esp8266/Arduino/issues/2162
https://github.com/esp8266/Arduino/issues/2607

Kameeno, 2018-02-19:
I'm trying everything, to use the 2.4.0 of the esp2866,  this is the i2c begin right now.. 
Wire.begin(5,4);
//Wire.setClock(10000L);
Wire.setClockStretchLimit(500);  but nothing resolve..  :(

LucAce, 2018-02-24:
Not sure which esp8266 board you are using so it is hard to say if you can even use those pins for I2C. I2C is an open collector bus protocol which basically means that the devices drive only to zero and let a board and/or IO pull-up return the bus to a 1. (https://en.wikipedia.org/wiki/Open_collector)  If the board you are using has pull-ups or pull-downs on the pins you are trying to use it may cause issues.

I have written my own CCS811 library and have an example sketch at https://github.com/LucAce/CCS811  The example is for a ESP8266 NodeMCU LUA CP2102 ESP-12E Internet WIFI Development Board and a Adafruit CCS811 Air Quality Sensor Breakout.  There is a png file that shows the wiring in the same directory.  I have been using a BME280, CCS811 and an OLED screen with the NodeMCU device for quite some time so I know they can be made to work.

I have found the CCS811 sensor to be kind of fussy.  I had another issue where reading the data register faster than the sample rate always caused the sensor to report no data available, which you would do if you were polling for data.  The solution I found was to make sure you polled the data ready at longer intervals than the configured CCS811 sample rate.

groubis, 2020-03-03:
I am also having issues with the sensor. I am using a feather huzzah esp8266 and clock stretching does not work as expected. Can anyone from adafruit support us on this?

ladyada, 2020-03-03:
try the library at https://github.com/LucAce/CCS811

groubis, 2020-03-03:
@ladyada tried without success... This is the result:

```
23:28:28.174 -> CCS811 Library Example
23:28:28.174 -> 
23:28:34.386 -> CCS811 Sensor Enabled:
23:28:34.386 -> Hardware ID:           0x81
23:28:34.386 -> Hardware Version:      0x12
23:28:34.386 -> Firmware Boot Version: 0x1000
23:28:34.386 -> Firmware App Version:  0x2000
23:28:34.386 -> 
23:28:34.386 -> CCS811 Startup In 1200 sec
23:28:34.386 -> 
```
and nothing else...

Tried the sensor with arduino uno r3 and works out of the box with your library. How can i adjust i2c clock stretching in feather huzzah esp8266?

Thanks,

ladyada, 2020-03-03:
you'd have to use Wire.setClockStretchLimit() - plz check google for some examples/documentation (we dont have any example code :)

groubis, 2020-03-03:
@ladyada the example in the library you suggested, already takes advantage of the Wire.setClockStretchLimit() method inside the setup() function. 

I tried with 500ms without success. Can you please advise if you can test the feather huzzah esp8266 with CCS811 and share the results? Maybe something is wrong with my device - but i do not believe so, since all other I2C sensors work out of the box.

Thanks,

groubis, 2020-03-04:
All examples i found online end-up without a solution. I guess i'll have to abandon the feather platform and continue with my arduino. Thanks, feel free to close this.

ladyada, 2020-03-04:
ok!

closed 2020-03-04 by ladyada

LucAce, 2020-03-07:
The CCS811 library I created was tested using version 2.4.1 of the Arduino ESP8266 library which is quite old at this point.  I'll update my build environment to the latest ESP8266 library (which is 2.6.3)  and test it again.  I don't have an Adafruit Huzzah ESP8266 on hand, just Adafruit HUZZAH32's and Wemos D1 Mini Pro's (V1.1.0) but the Wemos should be close enough.

If you have not already done so, make sure to try using the /RST (Active Low Reset) on the sensor between programmings and after a power cycle.  I have had issues with this sensor locking up between programmings or following resets of the microcontroller I was using.  It has been quite reliable after I started using a GPIO to control the reset of the sensor.