adafruit/Adafruit_CCS811 #9: ESP32, Adafruit CCS811 and deep sleep issues

krish2487, 2018-10-02:
Hello,
I am having a peculiar issue with the adafruit CCS811 breakout board and a ESP32 Wrover.
I am using the arduino framework and the breakout board from adafruit itself.

As it stands, the application presently starts, initializes a variety of sensors (BME680, CCS811, CO2 sensor etc) reads, records and displays the data and goes to deep sleep for a minute. It wakes up and does the same job all over again.

The CCS811 was a recent addition to the existing sensors and all the other sensors are working fine with no issues. However the CCS811 works only once and that too on a cold start. On wake up from deep sleep, the CCS811 does not seem to initialize at all. I have tried the application code with the Wakeup pin grounded and also connected to the ESP32, pulling it low on the ESP32 waking up from deep sleep and pulling it high before going to sleep again.

The example sketch provided with the library works just fine. I suspect it is a question of the ESP32 and the deep sleep mode. Also, the ESP32 performs a soft reset on wakeup and this reinitializes the sensor - I am guessing the sensor does not like this at all.

Is there anything I can do to fix this??
Any help is highly appreciated.

Thank you in advance. :-)

krish2487, 2018-10-02:
@ladyada Thank you. I ll try that and get back with my findings. :-)

Out of curiosity, would that not reset the sensor entirely and cause the sensor to never burn in and allow the readings to settle??
i.e it would always return a 400 ppm eco2 and 0 ppm tvoc since the sensor is always reset on wake up??

ladyada, 2018-10-02:
🤷‍♀️ no idea :D

krish2487, 2018-10-02:
@ladyada Thank you anyway.. I ll report back with my findings tomorrow.. Out of office for the day today..

krish2487, 2018-10-03:
okay. That seems to have done the trick. The sensor is repeatably working on wake up. However, I have to wait and confirm whether the reboot of the sensor is resetting the burn in phase as well.. 

@ladyada  thank you !! 👍

krish2487, 2018-10-03:
@ladyada  The readings are stuck at 400ppm for eCO2 and 0 ppb for tVOC.
I think that resetting the chip is also resetting the burn in phase for the sensor. Any other suggestions that I can try .

This is a stripped down version of the code I am running. Maybe you can try it on a ESP32 board and the CCS811 sensor to see if you can replicate the same issue??

```
#include "Adafruit_CCS811.h"
Adafruit_CCS811 ccs811;
void setup()
{
  //Adafruit CCS811 VOC Sensor
    pinMode(CCS811_WAKEUP, OUTPUT);
    delay(100);
    digitalWrite(CCS811_WAKEUP, LOW);
    delay(100);
    digitalWrite(CCS811_WAKEUP, HIGH);
    delay(100);
    if(!ccs811.begin()){
        Serial.println("Failed to start CCS811.");
      }
      else
      {
      //calibrate temperature sensor
      while(!ccs811.available());
      float temp = ccs811.calculateTemperature();
      ccs811.setTempOffset(temp - 25.0);
      sensorsInitialized += "CCS811; ";
      Serial.println("Initialized CCS811");
      }
}

void loop()
{

    Serial.println("Reading CCS811");
    while(!ccs811.available()){}
        Serial.println("Calculating Temperature");
        float temp = ccs811.calculateTemperature();
        if(!ccs811.readData()){
          Serial.println("Getting VOC values");
          payload["CCS811_CO2"]=(ccs811.geteCO2());
          payload["CCS811_TVOC"]=(ccs811.getTVOC());

    }
//Putting the ESP32 into deep sleep mode
esp_deep_sleep((uint64_t)60000000);
}
```

The "payload" is a json object where data from other sensors is logged as well and the entire object is logged to a server.

Thank you again!! :-)

ladyada, 2018-10-03:
hiya - we have no other suggestions

krish2487, 2018-10-04:
@ladyada  24 hours of continuous burn in and I observe only a constant 400 ppm eCO2 and 0 VOC levels.
It does not seem right.

Can I trouble you to replicate the above code and suggest what I might be doing wrong??

ladyada, 2018-10-04:
hiya we don't have any time to replicate the code and setup - this repo is just for the library - we know that CCS does have tech support for their sensors, perhaps they can assist. we've only used the sensor on an arduino uno

ladyada, 2018-10-04:
we'll close this because its more of a tech support question, not about the library. you can post in the adafruit customer forums if you have more questions!

closed 2018-10-04 by ladyada

bluesceada, 2019-09-07:
Here a comment despite its closed. How I handle it in the library, and keep CCS running through deep sleep:
I checked the code of this library, and it resets the CCS every time when you run "begin". Thus I am just running it when the device does not come out of deep sleep (in the following example, in case of the ESP8266). You can then always just read its values *after* deepsleep. And before deep sleep you do begin() and set its "driveMode" to only update results every 60sec, thus requiring at least 60sec deepsleep to get a result, but saving the most power. If you want something else check the other options (1sec,10sec,...) and read the official CCS811 datasheet from AMS to know what they mean.

`
rst_info * rstinfo = ESP.getResetInfoPtr();
// when NOT coming out of deepsleep:
if (rstinfo->reason != REASON_DEEP_SLEEP_AWAKE)
{
  if(!ccs.begin()){
    Serial.println("Failed to start/reset CCS811 sensor! Trying again later..");
  } else {
    ccs.setDriveMode(CCS811_DRIVE_MODE_60SEC);
    // you now need to wait at least 60sec before you would get a result,
    // so we wait until after deep sleep ...
  }
} else { // when waking up after deep sleep
  if(ccs.available()) {
    if(!ccs.readData()) { // valid data available
      CCS_eCO2 = ccs.geteCO2();
      CCS_TVOC = ccs.getTVOC();
      // ...... do whatever you want with the data ......
    }
  } else {
    Serial.println("CCS N/A .. resetting it!");
    ccs.begin();
    ccs.setDriveMode(CCS811_DRIVE_MODE_60SEC);
  }
}
ESP.deepSleep(62 * 1e6, WAKE_RF_DEFAULT); // sleep 62s (due to potential 2% clock speed difference, see CCS811 datasheet)
`

dadosch, 2020-04-10:
> Here a comment despite its closed. How I handle it in the library, and keep CCS running through deep sleep:
> I checked the code of this library, and it resets the CCS every time when you run "begin". [...]

unfortunately, the sensor doesn't return valid data for me then.

ccs811.available() is true, but .readData() returns false. If I read the register CCS811_ERROR_ID, it returns 255.

dannyyy, 2020-05-02:
@dadosch 
I was using the library of SpurkFun bit it lacks of the same incompatibility of using the ESP32 deep sleep. Due to your suggestion I switched to the library you mentioned but I experience a strange behavior with that.

The mode is set to every 60s. As long as I have my ESP32 connected via USB and the serial output is on, I get more or less meaningful data. If I place a tissue with aceton near the sensor it will  give high values immediately. But as soon as I power the ESP32 with my Li-Ion battery it takes several minutes from 400ppm/0ppb to a measured value. But if I place the same tissue next to the sensor, the TVOC is just 250ppb, which is way to low. 250ppm means low to very low TVOC. But it smells heavily meters away.

Do also met this observations? Could you publish your source code how you use the library together with the deep sleep functionality?

dadosch, 2020-05-02:
@dannyyy make sure your ESP is properly powered, meaning it has a large enough capacitor across the power line.

Here is how I did it: https://gist.github.com/dadosch/71253c0f5f52be0457114451374d28f6 There is not much different than in the libraries' examples.
Note that it might take up to 30 minutes to get the actual data, though this time is shorter with the newest firmware of the CCS811, but then no thermistor is supported anymore.

DeKasClimber, 2020-11-18:
@dadosch: do you have normal values? mine are increasing alle the time when using deep sleep. Both TVOC and CO2 values are very high.. with nearly nobody around.

With the basic sketch I have normal CO2 values varying from 410-550 ppm but in deep sleep values are increasing way beyond healthy levels..

Both sensors have a WEMOS D1 mini ESP8266 board with the CCS811 HDC1080 sensor

Deep sleep every 15 mins (on battery)
afbeelding
[image]

On power with basic sketch
[image]

dadosch, 2020-11-19:
@DeKasClimber 
did you make sure to run your sensor for at least 48 hours in the desired mode?
> CCS811 performance in terms of resistance levels and sensitivities will change during early life use. The change in resistance is greatest over the first 48 hours of operation. amsadvises customers to run CCS811 for 48 hours in the selected mode of operation to ensure sensor performance is stable

Also I would make sure to have the lastest firmware on the CCS811. Also it is likely your battery isn't powerful enough, make sure to add a large enough capacitor and make sure the battery's voltage doesn't get too low.

DeKasClimber, 2020-11-20:
Hi dadosch,

Thanks for the response.
The sensor is in the desired mode for weeks now, so the burn in took place if that is what you are referring to. Or do I need to create specific circumstances at startup?

The firmware of the sensor is 2.0.1. The battery is a 3.7 V Li-ion battery Panasonic NCR18650B (and similar - switching battery after a couple of weeks)

I have a 3.3V LDO (HT7533) with a 2000 mF and a 100nF capacitor combined.

It seems to me that values are starting from a new baseline on every wake-up of the board after deep sleep..

setup:
[image]