adafruit/Adafruit_CCS811 #10: Incorrect thermistor constants

nwf, 2019-07-02:
https://github.com/adafruit/Adafruit_CCS811/blob/7ff447cb883b9dc9380edbe00655635be45ba5ed/Adafruit_CCS811.cpp#L164 uses `CCS811_REF_RESISTOR` for the thermistor's resistance at 25C, but the schematic at https://learn.adafruit.com/assets/44639 indicates that this should be 10K, not 100K.

ladyada, 2019-07-03:
thermistor support was dropped from the firmware anyways :/

nwf, 2019-07-03:
Well that's unfortunate.  Did they give a rationale?

ladyada, 2019-07-03:
nope. but its definitely gone

closed 2019-07-03 by nwf

caternuson, 2019-07-03:
@nwf 100k looks correct?
[image]

@ladyada how does "dropped from firmware" affect `calculateTemperature()`?
https://github.com/adafruit/Adafruit_CCS811/blob/master/Adafruit_CCS811.cpp#L152

nwf, 2019-07-03:
@caternuson The 100K you so kindly circled corresponds to the use of `CCS811_REF_RESISTOR` on line **161**, which is correct.  I was referring to the use of it on line **164**, which is incorrect, as that should correspond to the resistance of the NTC device just to the right of the circled resistor, which, as you'll see in the schematic, is labeled as 10K, not 100K.