adafruit/Adafruit_CCS811 #12: ppm or ppb

mereshow, 2019-07-22:
Hi!

In your  [web](https://learn.adafruit.com/adafruit-ccs811-air-quality-sensor/overview) you indicate that TVOC is meassured in parts per billion (ppb) but in the examples you use parts per million (ppm). The [CCS811 sensor datasheet, page 9](https://cdn-learn.adafruit.com/assets/assets/000/044/636/original/CCS811_DS000459_2-00-1098798.pdf?1501602769) also indicates ppb. 
I don't see any TVOC conversion (ppb --> ppm) in your library, so I suppose that the examples display the incorrect units.

Thanks a lot for your work :-)

ladyada, 2019-07-22:
yep TVOC is in ppb and eCO2 is in ppm

[image]

ladyada, 2019-07-22:
where do you see TVOC mentioned in ppm, exactly?

mereshow, 2019-07-22:
Sorry! I didn't see the comma (that belongs to eCO2), the examples are correct! The TVOC units are not specified:

`Serial.print("ppm, TVOC: ");`

Sorry for wasting your time :-/

closed 2019-07-22 by mereshow