adafruit/Adafruit_CCS811 #13: Question setTempOffset()

esp13, 2019-09-09:
Hi,

In the example the temperature offset is set by comparing to 25.0

Why 25.0? Is it supposed to be the average room temperature, so instead of having the real temperature, it takes the 25.0 as average?

If yes, I could use my BME280 sensor to set the real offset? like this: `ccs.setTempOffset(temp - bme.readTemperature());`?

I don't really this temperature part. The CCS811 chip doesn't have any temperature sensor inside right? On my CCS811 breadboard I don't have any NTC temperature sensor neither I think (didn't see it). So how is this temperature moving?

Thank for reading and sorry for my poor english level.

ladyada, 2019-09-09:
it does not anymore - the old chips *used* to have temperature sensor support. it does not anymore, so this is sorta moot. :)

closed 2019-09-09 by ladyada

esp13, 2019-09-09:
Thank you for your fast answer!

(5)- So we can remove all setTempOffset() and calculateTemperature() call inside program?

(6)- What is now returning calculateTemperature()? Cause the returned data is moving.

ladyada, 2019-09-09:
yes dont use any of thsoe, they do not work with newer chips

bernardoafonseca, 2019-10-02:
Hi esp13,
did you understand the answer given by ladyada ?