adafruit/Adafruit_CCS811 #14: Sensor communication crashes randomly

jgrulich, 2019-10-22:
I'm testing the OLED Demo now. I have ESP core 2.5.2 and CCS811 lib 1.0.2 and the sensor communication fails after some time. The SDA line remains LOW and therefor blocks I2C buss. The OLED therefor stops working too. I've tested both 80MHz and 160MHz CPU freq.
I've tested than basic CCS811_test demo and it fails too. To bring it working again I need to hard reset the CCS811 board.

jgrulich, 2019-10-22:
Based on some info about the I2C stretching, I've tried to use this:
`   
    Wire.setClock(100000);
    Wire.setClockStretchLimit(500);
    Wire.begin();
    delay(100);

    if(!ccs.begin()){

`
But no change. I've tried to use another sensor and ESP boards too.

ladyada, 2019-10-22:
try on a non-ESP board. ESP has somewhat unusual I2C support

jgrulich, 2019-10-23:
Don't have native Arduino to test. But it's confirmed that it is the lib what is wrong. I've used this one instead and it works now for 12 hours. [https://github.com/LucAce/CCS811](url)

ladyada, 2019-10-23:
thanks, can you see what is different?

jgrulich, 2019-10-23:
Guess that the ccs.available() is wrongly detected, probably from the previous data read and than remains active by mistake. Than it crashes when tried to read new data too soon.
It's not so easy to identify, because the other lib is much more complex and the structure of commands is different. This ccs.available() command is not implemented there and it uses this instead:
`   while (!ccs.isDATA_READY()) {
      delay(1000);
      ccs.readStatusRegister();
   }
`

wangmings, 2020-07-27:
Have the above problems been solved？

JaySee56, 2022-02-18:
Have the same problem with a ProMini, 8MHz, IDE 1.8.13, Adafruit_CCS811.h