adafruit/Adafruit_CCS811 #21: no answer from sensor

rwussi, 2020-09-10:
There is no answer from the function ccs.begin().
When I disconnect wak from gnd or or scl from A5-Port of the arduino-board, I get the answer failed...
What is wrong?  

#include <Adafruit_CCS811.h>
Adafruit_CCS811 ccs;
void setup() {
Serial.begin(9600);
Serial.println("CCS811 test");
if (!ccs.begin()) {
  Serial.println("Failed to start sensor! Please check your wiring.");
  while(1);
  }
Serial.print("high");
//calibrate temperature sensor
while(!ccs.available());
float temp = ccs.calculateTemperature();
ccs.setTempOffset(temp - 25.0);
}
void loop() { 
...

ladyada, 2020-09-10:
hi, where did you buy the sensor?

rwussi, 2020-09-10:
I bought it from funduino!

ladyada, 2020-09-10:
great, please contact funduino for tech support - they have support engineers that will help you!

closed 2020-09-10 by ladyada