adafruit/Adafruit_CCS811 #25: ccs.readData() returns always 24 although measures looks correct

Waldmensch1, 2020-12-27:
Sensor: Breakout Board with CCS811 and HDC1080 (Temp/Hum) combined
Board: Heltec Lora32
IDE: PlatformIO

```
[env:ttgo-lora32-v1]
platform = espressif32
board = ttgo-lora32-v1
framework = arduino
monitor_speed = 115200
lib_deps = 
	closedcube/ClosedCube HDC1080@^1.3.2
	https://github.com/adafruit/Adafruit_CCS811
	olikraus/U8g2@^2.28.8
```

As readData never returns 0 I modified the code a bit to get results and see the retval

```
// global vars as used in UI thread for OLED
float temp = 0.0;
float hum = 0.0;
float eCO2 = 0.0;
float TVOC = 0.0;

void loop()
{
  temp = hdc1080.readTemperature();
  hum = hdc1080.readHumidity();
  ccs.setEnvironmentalData(hum, temp); // don't know whether this makes sense on any loop

  if (ccs.available())
  {
    int error = ccs.readData();
    //if(! error){
    Serial.print("eCO2: ");
    eCO2 = ccs.geteCO2();
    Serial.print(eCO2);
    Serial.print(" ppm, TVOC: ");
    TVOC = ccs.getTVOC();
    Serial.print(TVOC);
    Serial.print(" ppb, Temp:");
    Serial.print(temp);
    Serial.print(" °C, Hum:");
    Serial.print(hum);
    Serial.print(" %, Error:");
    Serial.println(error);
    //}
    //else{
    //  Serial.println(error);
    //while(1);
    //}
  }
  delay(500);
}
```

this is the output of the loop above. So there seems to be no error as the values look plausible.

```
eCO2: 702.00 ppm, TVOC: 46.00 ppb, Temp:23.44 °C, Hum:39.16 %, Error:24
eCO2: 700.00 ppm, TVOC: 45.00 ppb, Temp:23.52 °C, Hum:39.16 %, Error:24
eCO2: 699.00 ppm, TVOC: 45.00 ppb, Temp:23.54 °C, Hum:39.16 %, Error:24
eCO2: 694.00 ppm, TVOC: 44.00 ppb, Temp:23.45 °C, Hum:39.16 %, Error:24
eCO2: 694.00 ppm, TVOC: 44.00 ppb, Temp:23.53 °C, Hum:39.16 %, Error:24
eCO2: 694.00 ppm, TVOC: 44.00 ppb, Temp:23.54 °C, Hum:39.16 %, Error:24
eCO2: 694.00 ppm, TVOC: 44.00 ppb, Temp:23.54 °C, Hum:39.26 %, Error:24
eCO2: 694.00 ppm, TVOC: 44.00 ppb, Temp:23.49 °C, Hum:39.16 %, Error:24
eCO2: 692.00 ppm, TVOC: 44.00 ppb, Temp:23.59 °C, Hum:39.26 %, Error:24
eCO2: 682.00 ppm, TVOC: 42.00 ppb, Temp:23.49 °C, Hum:39.16 %, Error:24
eCO2: 688.00 ppm, TVOC: 43.00 ppb, Temp:23.45 °C, Hum:39.26 %, Error:24
eCO2: 688.00 ppm, TVOC: 43.00 ppb, Temp:23.41 °C, Hum:39.36 %, Error:24
eCO2: 679.00 ppm, TVOC: 42.00 ppb, Temp:23.52 °C, Hum:39.36 %, Error:24
eCO2: 679.00 ppm, TVOC: 42.00 ppb, Temp:23.52 °C, Hum:39.36 %, Error:24
eCO2: 692.00 ppm, TVOC: 44.00 ppb, Temp:23.49 °C, Hum:39.36 %, Error:24
eCO2: 688.00 ppm, TVOC: 43.00 ppb, Temp:23.54 °C, Hum:39.36 %, Error:24
eCO2: 688.00 ppm, TVOC: 43.00 ppb, Temp:23.47 °C, Hum:39.26 %, Error:24
eCO2: 699.00 ppm, TVOC: 45.00 ppb, Temp:23.47 °C, Hum:39.26 %, Error:24
eCO2: 697.00 ppm, TVOC: 45.00 ppb, Temp:23.44 °C, Hum:39.26 %, Error:24
eCO2: 692.00 ppm, TVOC: 44.00 ppb, Temp:23.52 °C, Hum:39.26 %, Error:24
```

ladyada, 2020-12-27:
hihi we just tested this library and its ok. please contact the place you got the sensor from for tech support.

closed 2020-12-27 by ladyada

Waldmensch1, 2020-12-27:
Ahh okay, so it seems that there are different CCS811 in wildlife. Strange, as it responses on 0x5A with correct values. Wasn't aware that the errorcode is generated by sensor itself. Good to know. I also can live with above Workaround.